philippe44/squeezelite-esp32 #5: Add support for display

sle118, 2019-06-04:
Espressif has made littlevgl available of the platform.

https://github.com/espressif/esp-iot-solution/blob/master/documents/hmi_solution/littlevgl/littlevgl_guide_en.md

daduke, 2019-08-16:
for the last 3 hours I tried setting up 'esp iot solution' and run a display test example. It does **not** compile using the 'new' tool chain (gcc 8.2). I can't find the 5.2 tool chain for Linux any longer and even if I could, I don't think that's the route to go. Any hints?

philippe44, 2019-08-17:
you're talking about the specific display project, not about the squeezelite-esp32, right?

daduke, 2019-08-17:
correct. squeezelite-esp32 compiles but stutters (you remember), but the "iot solution" which includes littleevgl throws hundreds of errors. I opened an issue https://github.com/espressif/esp-iot-solution/issues/51, let's see what they say.

daduke, 2019-08-17:
ok I finally managed to install IDF 3.2 with gcc 5.2. Now the littleevgl do compile, but squeezelite-esp32 doesn't (nvs stuff unknown). Seems we can have either littleevgl or squeezelite-esp32, but not both. What a pain...

sle118, 2019-09-03:
Let's keep this enhancement thread dedicated to the intended topic, please :)

sle118, 2020-01-27:
Basic display for commonly available i2s oled screens has been implemented in sle118's repository, so closing this one.

closed 2020-01-27 by sle118