philippe44/squeezelite-esp32 #9: unable to make

marrog86, 2019-08-29:
Getting failure while making becouse make and project.mk not existing. 

``squeezelite-esp32-refactor-BT-I2C$ make defconfig
Makefile:8: /make/project.mk: Datei oder Verzeichnis nicht gefunden
make: *** Keine Regel, um „/make/project.mk“ zu erstellen.  Schluss.
``
thanks

sle118, 2019-08-29:
Could you describe your build environment? This error could be caused by something as simple as not having IDF_PATH environment variable configured.

marrog86, 2019-08-29:
Thanks for your hint. After installing correctly compile works. will try flashing it in one week. 

Other questions: Basically, this is the original squeezebox emulator code ported to Esp32? How are handled all the commands which normally used in console. 

and: Is there a posibility to connect generic dac or pwm on uart?

And the most important question: comparison of your other esp-espressiv with arduino library, which is mor stabile? what the core differences?

Thanks for help and your super projects, i like to bild my own squeezbox wifi receaver.

philippe44, 2019-08-29:
Yes, it is squeezelite but I had to modify if a lot to make it work on a small, non-really posix, embedded system. Having said that, I've preserved full compatibility with Windows/Linux so basically you could take this squeezeliteesp32 and use it as the main squeezelite version moving forward. To a large extend, this is what the squeezelite-core project is.
The command line is set in the NVS of the esp32 and is very similar to squeezelite cmd line.
I don't use arduino library, they probably abstract too much the esp32 for what I wanted to do. But I think the esp-idf is the base for both projects
you can connect generic dac on i2s, not on uart.
For a full project with hardware, look at my squeezeamp project

sle118, 2019-09-03:
To add on top of what Philippe44 wrote, have a look at the SqueezeAmp project and keep in mind that Philippe44 made the design very modular. You could build a simple device with an esp32+spdif and that would be the bare minimum to get all the features of the code base. 

You can also visit the forum for official discussions:
https://forums.slimdevices.com/showthread.php?110926-pre-ANNOUNCE-SqueezeAMP-and-SqueezeliteESP32&p=949517#post949517

closed 2019-09-03 by sle118

marrog86, 2019-09-03:
I saw the SqueezeAmp Project with interrest. But i have not the posibility vor making electronic boards, so this gets complicated. Also its complicated for me finding self made failures during assembling because of missing equippment. Thats the reason why I prefere using pre-assambled boards hoping that they are working out of the box how they should, and for me to debug staying only a small amount of connections.