philippe44/squeezelite-esp32 #15: Implement basic authentication for HTTP server

sle118, 2019-09-06:
Basic authentication is going to be needed if we want to allow OTA updates and avoid unwanted software being injected. Here are some points to consider

- Default access point name is "squeezelite-9999" where 9999 is derived from the cpu's eFuse MAC
- Default user and password to something as simple as admin/admin-9999 where 9999 are the same letters as the ones for the access point name.  This makes it easy to know what the password of the device is on the initial configuration
- During initial configuration, user must change the defaults before they can configure WiFi 
- No need for multi user; this would add complexity and bring low value
- User need to change BOTH the user name and the password

daduke, 2019-09-06:
does http_server support https? If not, the pw will go over the air in plain text and we can save us the hassle

sle118, 2019-09-06:
WiFi on initial config is encrypted. Not perfect, but not quite clear text.

closed 2020-08-28 by philippe44