philippe44/squeezelite-esp32 #17: Compile Problems

marrog86, 2019-09-09:
I tried to Compile with "make flash" how described in the main Page and got following error: 

`CC build/cmd_nvs/cmd_nvs.o
/home/martin/Schreibtisch/DEV/ESP/squeezelite-esp32-refactor-BT-I2C/components/cmd_nvs/./cmd_nvs.c:30:5: error: unknown type name 'nvs_type_t'
     nvs_type_t type;
     ^

/home/martin/Schreibtisch/DEV/ESP/squeezelite-esp32-refactor-BT-I2C/components/cmd_nvs/./cmd_nvs.c:35:7: error: 'NVS_TYPE_I8' undeclared here (not in a function)
     { NVS_TYPE_I8, "i8" },
       ^

/home/martin/Schreibtisch/DEV/ESP/squeezelite-esp32-refactor-BT-I2C/components/cmd_nvs/./cmd_nvs.c:36:7: error: 'NVS_TYPE_U8' undeclared here (not in a function)
     { NVS_TYPE_U8, "u8" },
       ^

/home/martin/Schreibtisch/DEV/ESP/squeezelite-esp32-refactor-BT-I2C/components/cmd_nvs/./cmd_nvs.c:37:7: error: 'NVS_TYPE_U16' undeclared here (not in a function)
     { NVS_TYPE_U16, "u16" },

....
....
...

cc1: some warnings being treated as errors
/home/martin/esp/git/esp-idf/make/component_wrapper.mk:273: die Regel für Ziel „cmd_nvs.o“ scheiterte
make[1]: *** [cmd_nvs.o] Fehler 1
/home/martin/esp/git/esp-idf/make/project.mk:450: die Regel für Ziel „component-cmd_nvs-build“ scheiterte
make: *** [component-cmd_nvs-build] Fehler 2

`

I made the configuration with defaultconfig and in Menuconfig only changing the com port. Oter options are on default.

daduke, 2019-09-10:
that happened to me if I didn't use esp idf 4. Note that this project requires the rather unusual combination IDF 4 + gcc 5.2

marrog86, 2019-09-11:
I installd idf4, but still with errors. 
I wasn't unable to downgrade gcc to 5.2 on LInux Mint LMDE because of missing sources. I tried to add debian oldstabile without success. 
Somebody can give me a hint how to install old gcc on an well updated Linux Mint?

daduke, 2019-09-12:
I'm not talking about your Linux system's gcc (which will produce amd64 binaries) but the xtensa toolchain for the ESP. You'll basically have to frankenstein an IDF 3.2 gcc 5.2 into IDF 4... not sure there's tutorials for this, for me it was a manual process.

marrog86, 2019-09-13:
yes, i made make defconfig

marrog86, 2019-09-13:
I noticed: My controller is an ESP32-WROOM which has not external Ram. I ordered an other one.

sle118, 2019-09-13:
in case someone else has build issues, the first thing to do is to try using one of the build scripts from the repo as their "sdkconfig" file.  This will ensure that at least the options are the same as our baseline.

Also, note that pre-built binaries are now available at
https://1drv.ms/f/s!Ajb4bKPgIRMXmwzKLS2o_GxCHRv_

closed 2019-09-13 by sle118