philippe44/squeezelite-esp32 #25: Using internal DAC

marrog86, 2019-09-27:
During waiting for an I2S dac, i wanted to try it with internal one. 

Espressif said on 
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/i2s.html
you have to send NULL on i2s_set_pin(i2s_num, NULL) for enabling the internal dac
I changed the code on output_i2c line 313 to:
` i2s_set_pin(CONFIG_I2S_NUM, NULL);//&pin_config); //chanched here for DAC output`
But it isn't working. 

The documentation says this whil configure dac to pin 25 & 26, but nothing happens. 
Thanks for help.

marrog86, 2019-09-30:
I also had to disable the I2S_COMM_FORMAT_I2S|  parameter. 
Now i was able to verify that the cables are ok, but the audio is only cracking sound. 

What could be the reason?
here the log outputs: 

`squeezelite -o I2S -b 500:2000 -d all=debug -m ESP32

D (13355) squeezelite_cmd: Parameters:
D (13355) squeezelite_cmd:      squeezelite
D (13355) squeezelite_cmd:      -o
D (13355) squeezelite_cmd:      I2S
D (13365) squeezelite_cmd:      -b
D (13365) squeezelite_cmd:      500:2000
D (13365) squeezelite_cmd:      -d
D (13375) squeezelite_cmd:      all=debug
D (13375) squeezelite_cmd:      -m
D (13375) squeezelite_cmd:      ESP32
D (13385) squeezelite_cmd: assigning parm 0 : squeezelite
D (13385) squeezelite_cmd: assigning parm 1 : -o
D (13385) squeezelite_cmd: assigning parm 2 : I2S
D (13395) squeezelite_cmd: assigning parm 3 : -b
D (13395) squeezelite_cmd: assigning parm 4 : 500:2000
D (13405) squeezelite_cmd: assigning parm 5 : -d
D (13405) squeezelite_cmd: assigning parm 6 : all=debug
D (13425) squeezelite_cmd: assigning parm 7 : -m
D (13425) squeezelite_cmd: assigning parm 8 : ESP32
D (13425) squeezelite_cmd: Starting Squeezelite Thread
D (13435) squeezelite_cmd: Back to console thread!
I (13435) squeezelite_cmd: Waiting for WiFi.
D (13445) squeezelite_cmd: Number of args received: 9
D (13445) squeezelite_cmd: Values:
D (13445) squeezelite_cmd:      squeezelite
D (13455) squeezelite_cmd:      -o
D (13455) squeezelite_cmd:      I2S
D (13455) squeezelite_cmd:      -b
                                                                                                                                                                                                                                             D (13465) squeezelite_cmd:      500:2000
D (13465) squeezelite_cmd:      -d
D (13465) squeezelite_cmd:      all=debug
D (13475) squeezelite_cmd:      -m
D (13485) squeezelite_cmd:      ESP32
D (13485) squeezelite_cmd: Starting Squeezelite runner Thread
I (13495) squeezelite_cmd: Calling squeezelite
I (13495) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
[00:00:12.178298] stream_init:370 init stream
[00:00:12.179085] stream_init:371 streambuf size: 512000
[00:00:12.190123] output_init_embedded:52 init device: I2S
[00:00:12.190885] output_init_common:351 outputbuf size: 2048000
[00:00:12.202193] output_init_common:375 idle timeout: 0
[00:00:12.203262] output_init_common:427 supported rates: 192000 176400 96000 88200 48000 44100 32000 24000 22050 16000 12000 11025 8000 
squeezelite-esp32> [00:00:12.215736] output_init_embedded:64 init I2S/SPDIF
[00:00:12.226725] output_init_i2s:310 Initializing I2S mode normal with rate: 44100, bits per sample: 16, buffer frames: 512, number of buffers: 12 
I (13565) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=12
I (13565) I2S: APLL: Req RATE: 44100, real rate: 22049.982, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 5644795.500, SCLK: 705599.437500, diva: 1, divb: 0
[00:00:12.254226] output_thread_i2s:441 Output state is 0
[00:00:12.265915] output_thread_i2s:496 Restarting I2S.
[00:00:12.265915] output_init_embedded:70 init completed.
[00:00:12.277788] decode_init:157 init decode
[00:00:12.278915] register_alac:547 using alac to decode alc
[00:00:12.279942] register_helixaac:657 using helix-aac to decode aac
[00:00:12.290952] register_vorbis:378 using vorbis to decode ogg
[00:00:12.292003] register_opus:320 using opus to decode ops
[00:00:12.303022] register_flac:306 using flac to decode flc
[00:00:12.304175] register_pcm:483 using pcm to decode aif,pcm
[00:00:12.315274] register_mad:417 using mad to decode mp3
[00:00:12.316282] decode_init:210 include codecs:  exclude codecs: 
[00:00:12.320910] discover_server:791 sending discovery
[00:00:12.320910] output_thread_i2s_stats:576 Heap internal:127816 (min:126368) external:1484760 (min:1484760)
[00:00:12.517209] discover_server:802 got response from: 192.168.8.16:3483
[00:00:12.518957] slimproto:890 connecting to 192.168.8.16:3483
[... 75 lines cut ...]
[00:00:15.525080] output_thread_i2s:441 Output state is 2
[00:00:15.524868] process_strm:333 unpause at: 0 now: 15529
[00:00:15.538020] sendSTAT:191 STAT: STMr
[00:00:15.554424] process:525 audg
[00:00:15.555326] process_audg:437 audg gainL: 65536 gainR: 65536 adjust: 1
[00:00:15.560065] set_volume:80 setting internal gain left: 65536 right: 65536
[00:00:16.043512] sendSTAT:191 STAT: STMt
[00:00:16.053156] process:525 strm
[00:00:16.054116] process_strm:276 strm command t
[00:00:16.056269] sendSTAT:191 STAT: STMt
[00:00:17.088050] sendSTAT:191 STAT: STMt
[00:00:18.128803] sendSTAT:191 STAT: STMt
[00:00:19.144834] sendSTAT:191 STAT: STMt
E (20645) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (20645) task_wdt:  - IDLE1 (CPU 1)
E (20645) task_wdt: Tasks currently running:
E (20645) task_wdt: CPU 0: IDLE0
E (20645) task_wdt: CPU 1: output_i2s
[00:00:20.162585] sendSTAT:191 STAT: STMt
[00:00:21.057032] process:525 strm
[00:00:21.057926] process_strm:276 strm command t
[00:00:21.059399] sendSTAT:191 STAT: STMt
[00:00:22.096448] sendSTAT:191 STAT: STMt
[00:00:23.125666] sendSTAT:191 STAT: STMt
`

marrog86, 2019-10-03:
Pleas only tell mi if it should be possible no make this with the internal DAC, so i woll not loose time.  
i am now able to use ist, but only with changing sample rate to 20000 and the quality is realy poor (no idea if it is the sample rate or other problem)

Thanks

philippe44, 2019-10-03:
I don’t know if this is possible or not, I’ve never tried so you know as much as I do

sle118, 2019-10-11:
I have ruled out the use on internal dac early on during the porting effort. First, the internal dac are 8 bits, which means low quality for music. Second, you would need to split the decoded stream into individual left and right channels for this to work. Typical i2s DAC will extract the channels for you.

your only option to reduce component count is to feed an SPDIF output into an amplifier.

Given how cheap the I2S headphone codec boards are (<$10), there really isn't an interest to support this configuration. You are welcomed to try and report back if you think there is some potential.  

thank you!

closed 2019-10-11 by sle118

philippe44, 2019-10-11:
Oh BTW you can use the esp32-a1s made by hilink which has the wrover pinout and integrates a DAC. Go on slimdevice forum, somebody has made it work