philippe44/squeezelite-esp32 #28: Squeezelite and Lyriat board

kheperV3, 2020-08-23:
I am trying to install Squeezelite on a Lyriat board.
I think that the biggest problem is to handle the ES8388 chip (vs the AC101 on the SqueezeAmp board).
I think there would be  2 possibilties:
- to use the nvs parameter "dac_controlset" but I'm afraid it's very complex and difficult...
- to write a specific handler ((on the AC101 model) but in this case I will have difficulties in intoducing it in your squeezelite...
What do you think about this ?
Many thanks!...

(this is not an issue but a request for an advice!)

philippe44, 2020-08-23:
It depends what the ES8388 needs in term of I2C config. Do you know? 
BTW, SqueezeAMP is a TAS5756/54/80, not a AC101

kheperV3, 2020-08-24:
Yes I know it well enough (I wrote a handler for another app)
(0x10 is the i2c chip address, up to 45 registers to be managed via i2c - much less in a simple use...)
Could vous give me your opinion on this problem?
Thanks!

pgu-swir, 2020-08-24:
if all you need to have the basics of the chip is a few commands, then dac_controlset is the way to do). If it requires *a lot* of handling, then you'll have to take one of the drivers as an example and go the C route

philippe44, 2020-08-24:
How many for a simple use? you could use dac_controlset, it's not complicated. Otherwise you'll have to go the C route

kheperV3, 2020-08-25:
Hi Philippe,
I'm trying the "dac_controlset" way..
23 register writes for "init"
1 for "poweron"
1 for poweroff"

For the moment, it does not work...
I'm trying to check the syntax...
Any other idea ?
(by the way, I work from the "generic I2S & BT" model, that's right?)
Thanks!

kheperV3, 2020-08-31:
Hi Philippe,
Today I worked on my Lyriat board using these nvs parameters :
dac_config => bck=5,ws=25,do=26,sda=18,scl=23,i2c=16,port=0
dac_controlset => {"init":[{"reg"=25,"val"=4},{"reg"=2,"val"=0},{"reg"=8,"val"=0}{"reg"=4,"val"=192},{"reg"=0,"val"=18},{"reg"=1,"val"=0},{"reg"=23,"val"=24},{"reg"=24,"val"=2},{"reg"=38,"val"=9},{"reg"=39,"val"=144},{"reg"=42,"val"=144},{"reg"=43,"val"=128},{"reg"=45,"val"=0},{"reg"=27,"val"=0},{"reg"=26,"val"=0},{"reg"=46,"val"=33},{"reg"=47,"val"=33},{"reg"=4,"val"=12},{"reg"=25,"val"=0}],"poweron":[{"reg"=2, "val"=0}],"poweroff":[{"reg"=2,"val"=3}]}

And I watched the I2C signals SDA ans SCL using a logic analyser.
=>It seems that there is no I2C transaction.
What do you think about that ? 
Syntax errors ?
False values? 

Many thanks
Louis

philippe44, 2020-08-31:
what do you have in traces?

philippe44, 2020-08-31:
And you're missing a few comma in the payload and a validator like this one https://jsonlint.com/

Use that example { "init": [ {"reg":41, "val":128}, {"reg":18, "val":255} ], "poweron": [ {"reg":18, "val":64, "mode":"or"} ], "poweroff": [ {"reg":18, "val":191, "mode":"and"} ] }

kheperV3, 2020-08-31:
Thanks for the json validator.
It seems that I have a lot of syntax errors !
I will correct all this from next week (short vacation...)
I'll keep you informed...

kheperV3, 2020-09-15:
Hi Philippe!
It works now!
1- I corrected the syntax of "dac_controlset"
2- But I had to add 2 lines in your code to create the MCLK clock (for the ES8388) from the CLK_OUT1 signal (pin23/gpio0 of the ESP32)

I wonder if this could not be done with an NVS parameter (existing or to be created) ?
What do you think about this?
Thanks!
Louis

pgu-swir, 2020-09-16:
> Hi Philippe!
> It works now!
>
Cool
> 2- But I had to add 2 lines in your code to create the MCLK clock (for the ES8388) from the CLK_OUT1 signal (pin23/gpio0 of the ESP32)
> 
Can you show what you had to do?

philippe44, 2020-09-16:
Can you share what you did?

kheperV3, 2020-09-16:
What I did:
1- NVS params
dac_config => bck=5,ws=25,do=26,sda=18,scl=23,i2c=32,model=I2S
dac_controlset => {"init": [{ "reg": 0, "val": 128 }, { "reg": 0, "val": 0 }, { "reg": 25, "val": 4 }, .... ( about 20 items)
set_GPIO => 21=amp
2- in file "esp_app_main.c"
at line 368 (after "config_init();") I added this:
     // GPIO0 => CLK_OUT1
      PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
      WRITE_PERI_REG(PIN_CTRL, READ_PERI_REG(PIN_CTRL)& 0xFFFFFFF0);

philippe44, 2020-09-16:
oh but can't you just not use MCLK (register 43, bit4)?

kheperV3, 2020-09-17:
I have already tried this. But it does not work on the Lyriat board (why?...)

dtorner, 2021-03-21:
Hey @kheperV3 would you mind to share the config you used to get the lyrat board working? I'm trying to figure it out by myself, but I'm having problems... Thanks in advance.

kheperV3, 2021-03-22:
Hi dtorner
You have to do what is in my September post .
And for "dac-controlset" you must put the following values (respecting the json format)
R00 = 00
R25 = 04
R01 = 50
R02 = 00
R08 = 00
R04 = c0
R00 = 12
R01 = 00
R23 = 18
R24 = 02
R38 = 09
R39 = 90
R42 = 90
R43 = 80
R45 = 00
R27 = 00
R26 = 00
R02 = f0
R02 = 00
R29 = 1c
R04 = 30
R25 = 00
R46 = 21
R47 = 21
I hope this will be useful for you ...

dtorner, 2021-03-22:
Thanks you very much @kheperV3, I'll try to get it working using that info.

Mikescops, 2022-01-12:
Hey @kheperV3, thanks for the work on LyraT.
I'm just getting into playing with this board and the squeezelite project.

I followed your post here https://github.com/philippe44/squeezelite-esp32/issues/28#issuecomment-693301216 but I'm struggling with the last part with the dac_controlset setting.

I get the idea of converting the different registers to a JSON format but how does it fit in the different actions possible (init/poweron/...)

I'm curious to know (and eager to learn) how you went to have everything working :smile:

philippe44, 2022-01-13:
This https://github.com/sle118/squeezelite-esp32/blob/master-cmake/README.md#daci2s section of the README tells you how to use dac_controlset

Mikescops, 2022-01-13:
thanks @philippe44, I've already looked at how to use the dac_controlset and I get it but what I don't understand is how to find the expected registers that I need to set (sorry if it's a stupid question)

philippe44, 2022-01-13:
Don't worry about stupid or not stupid question :smile:. Now if your board as an ES8388 codec, support is now built-in, you just have to set model=ES83888 (see README) and set the proper gpio in the dac_config and you are good to go!

Mikescops, 2022-01-13:
Damn, how could I miss that :sweat_smile: Thanks a lot! It's working perfectly :tada: 

Ok, now I have to take some time to understand it all but that's a start, thanks again :)

philippe44, 2022-01-13:
Glad to help (un Français peut-être?)

Mikescops, 2022-01-13:
oui ! :fr:

philippe44, 2022-01-13:
Tu peux nous retrouver sur notre gitter https://gitter.im/SqueezeAmp/community?source=orgpage (ça ne parle pas que de SqueezeAMP)

gnkarn, 2022-04-10:
Hi , im trying to get the lyrat firmware  flashed, and i tried  to make the configuration working by compiling with VS code studio ( and the  extensions) , with no luck after spending lot of time , so my question are : is there a binar i could flash , to get it initially working?, is it possible to flash it from a MAC?, si there a step by step procedure to get the environment correctly set for VSC?  , thanks a lot and appreciate any help .

philippe44, 2022-04-10:
Please go there https://github.com/sle118/squeezelite-esp32